NoEngineerHere/TeensyELS #19: ESP32 fork

martinlong1978, 2025-01-05:
Hi @jh247247 and @NoEngineerHere   Not really an issue, but YouTube is for some reason blocking my comments. Just wanted to say check out my ESP32 fork martinlong1978/TeensyELS 

I'm working on it so that it will build for both architectures, so it should be completely mergable once it's in a good enough place. It's starting to get close to competion.

jh247247, 2025-01-06:
Dude thats awesome! Thanks so much for your work! I had a look at your fork and I'm very impressed!
Hopefully my structure wasn't too difficult to work with, I know it has plenty of rough edges but from the looks of it you didn't have any trouble.

I also noticed you added support for a tft display and multiplexed keypads along the way, very cool!

When you think its ready I'll have a look again but I'm so appreciative you took the time to work on this - you've done way more than I've been able to the past few months.

Btw I did reply to your comments on youtube so I have no idea what is going on 😬 I'll put a screenie here:
[image]

martinlong1978, 2025-01-06:
Yeah, I saw those replies - thank you. I tried to comment again though to point you at my fork, and it just kept disappearing. 

Anyway, it's still a bit rough around the edges. I'd probably want to separate the screen types into different files/classes, and give the button classes a common base type, for example, but it's functional for now. Just waiting on my rotary encoder from China before I can do much more.

jh247247, 2025-01-07:
Turns out I don't get a notification unless mentioned - sorry! @martinlong1978 
We've been using "solderable breadboards" - essentially veroboard with a breadboard layout, there were some murmurings about making a custom PCB early on but that didn't really eventuate...

There was an `rtf` file that @NoEngineerHere uploaded on a branch that never merged, not a schematic but better than nothing? https://github.com/NoEngineerHere/TeensyELS/tree/Documentation/Documentation
But yes a couple of n-channel mosfets on one of those cheap level shifting breakout boards is more than enough for the external connections to the teensy.  Would be cool to break out the tiny cnc I bought for making PCBs to actually make PCBs for once!

I was planning on adding support for the ILI9341 for a screen such as: https://www.aliexpress.us/item/3256806731848791.html in the future, so maybe a generic UI library (with touch support) may be useful in the future if we're adding support for higher resolution TFT displays

martinlong1978, 2025-01-08:
Thanks @jh247247 for that info. I'll take a look at that later. 

I probably will get some boards sorted anyway. They're so cheap on JLCPCB I generally get them for every project. It'll also give me an easy pathway back to the Teensy if the ESP32 isn't cutting it. 

The TFT library I added should actually support a whole bunch of TFT displays. I haven't checked the specific one you linked, but it's a fairly big list. (In just checked, it's on the list). I also stayed away from capacitive touch for the workshop... I don't like the idea of grease and oil and greasy fingers on a sensitive capacitive display. But, you know... YMMV

But, yes, in essence it would be good to tidy up the UI stuff and add some kind of interface to make adapting the UI to different resolutions and different input methods would be a definite win.

martinlong1978, 2025-01-21:
@jh247247 Closing this now. I'll send a pull request once I've done some real-world testing on my fork.

closed 2025-01-21 by martinlong1978