NoEngineerHere/TeensyELS #20: ratio is backwards.

martinlong1978, 2025-01-12:
Hi @jh247247 

I think the ration is being caculated backwards. I reckon it hasn't shown itself as a bug, because you've been using the same encoderPPM and motor PPM so it's constant.

float Leadscrew::getRatio() { return m_ratio * (float)(((float)leadAxisPPR) / ((float)motorPulsePerRevolution)); }

I've thought about this, done some mental gynastics and then thought some more. I'm pretty sure leadAxisPPR and motorPulsePerRevolution are the wrong way around. I think it should be motor pulses per encoder pulse. 

I spotted this because I changed the microsteps on my motor, and when I changed the stepper PPR to match it had the opposite effect. 

Let me know what you think.

jh247247, 2025-01-13:
Hey! I'm happy to have it swapped around - I'm pretty sure I may have flipped it once before in desperation. I do have a different ratio of encoder pulses per rotation to motor pulses per rotation (2000 , 1600) .
If it makes sense to you go ahead! These mental ratios really did my head in the first time around so I'm not surprised I got a few things critically wrong (but somehow right because the ratios work out on my lathe with weird values 😅 )

martinlong1978, 2025-01-21:
@jh247247  Commited to my fork. I'll close this now.

closed 2025-01-21 by martinlong1978