NoEngineerHere/TeensyELS #21: ELS_GEARBOX_RATIO not accounted for in pitch calculations.

martinlong1978, 2025-01-12:
@jh247247 So, the ELS_GEARBOX_RATIO appears to be the gearbox between the stepper and the leadscrew, such that a value of 2 here means 2 revolutions of the stepper will result in 1 revolution of the lead screw. Great. That make sense where we're passing in ELS_LEADSCREW_STEPPER_PPR* ELS_GEARBOX_RATIO into the constructor of the leadscrew. 

However, it doesn't seem to be accounted for in the ELS_LEADSCREW_STEPS_PER_MM macro. 

[image]

This is then used in what looks like the accelleration calculations. 

I also notice you were using a value of 95. Seems like a very high ratio. 

(btw, I'm not looking for you to fix these issues I'm raising. I'm fixing them in my branch. I just want you to confirm my findings)

jh247247, 2025-01-13:
Yep you're correct here too - I initially added this into the constructor for the `Leadscrew` but having it taken into account here makes way more sense (Sorry! - I was under a time crunch for final implementation!)

I also have no idea why the number I landed on for the ratio was so high, I did some calibration to ensure that when 1mm was set as the pitch one rotation actually moved that much. (My ratio's should be closer to about 1 since I have a 1:1 ratio on my belt and a ~0.9? ratio on the gearbox, I found much better performance by setting it in the highest setting since at high speeds the stepper can't keep up)

jh247247, 2025-01-13:
I actually even called out how the gearbox ratio is in a bad state in my "mad dash" calibration PR haha https://github.com/NoEngineerHere/TeensyELS/pull/17

martinlong1978, 2025-01-13:
@jh247247 cool. I'm not going to reply to each issue, but thanks for all the feedback. I'll apply fixes in my branch. It should be ready for merge fairly soon.

martinlong1978, 2025-01-21:
@jh247247  Commited to my fork. I'll close this now.

closed 2025-01-21 by martinlong1978