NoEngineerHere/TeensyELS #23: PCB design

martinlong1978, 2025-01-18:
Hi @jh247247  @NoEngineerHere  I've done some PCB design work. This incorporates the option to use either microcontroller. It uses the keypad layout and adds a UI rotary encoder for selecting pitch. 

I've also wired up enable and alarm from the stepper driver to give visual feedback and the capability to reset it. 

The spindle encoder can be an NPN type or voltage signal, by adjusting the components soldered to the board. 

It does use some SMT components, but they're fairly easy to solder, 0806 resistors and some BSS138 mosfets. This eliminates the need for a level shifter board. I've also generally used external pull-up resistors rather than relying on the on-chip pull resistors, which can fail. 

The interface to the stepper is an RJ45 connector - you can simply plug in a patch cable, and then strip the other end and wire the pairs up to the stepper. 

It's still an early draft, but you can take a look here: https://365.altium.com/files/A7A283D5-AF37-43DF-8B6F-CDB23BD8D74F

jh247247, 2025-01-22:
Awesome! I can't open the PCB due to some errors but it's cool to hear you've made progress on the hardware side!

Would be really cool if we could use something like kicad instead of altium, I've used it before but it's a bit rich for my blood nowadays!

martinlong1978, 2025-01-31:
@jh247247 I've been meaning to learn KiCAD for years, but kept on using CircuitMaker because I know it and so it's low friction. About time I took the time to learn it. 

So... I've just committed and pushed KiCAD designs to my branch. I don't know if you need the footprints and symbols I added separately. Anyway, have a look and let me know what you think.

martinlong1978, 2025-02-25:
@jh247247 

[image]