NoEngineerHere/TeensyELS #24: Jog seems broken

martinlong1978, 2025-02-24:
@jh247247  Hi

I've made a bunch of progress. Go my boards from JLCPCB, printed a case, buttons, etc, and done a whole load of debugging, testing refactoring etc. 

A few highlights: 
  - self updating from a URL over wifi on ESP
  - remote console debugging on ESP
  - ESP now uses RMT hardware for sending pulses == lightning fast
  - ESP also uses registers for setting DIR pin == lightning fast
  - Added rotary encoder for setting pitch, with an RGB blinky light in the middle
  

Anyway... It looks like jog is broken in your most recent "boj-position-consumption-fix" branch. JOG never gets set to motionMode, and the JOG_PULSE_DELAY or JOG_SPEED never gets used. Did you switch to some other way that JOG operates, because I can't quite figure it out. I see that it broke shortly after you switched it from working when the button is held down, to jogging when it is single clicked. I see that it should jog straight to the endstop, but this doesn't seem to happen. Also, it leaves me a bit perplexed as to how end-stops should be set using this new UX - how do I position the tool without being able to control the jog start/stop?  Does it require reversing the spindle? (which seems cumbersome when we have buttons). 

Happy to fix it, but would like your insight as to what is intended behaviour, so I can figure out the gap. Thanks.

jh247247, 2025-02-25:
Hey!
That's great! Glad to hear all the progress you've made so far!
Would love to have a pull request if you have it in a working state, so we can all contribute and have things up to date here too!

Yeah with the jogging I know I broke it at some point. I wanted to change it from a hold down to move, to jog to either end stop.

So the behavior should be:
- Travel to start under power
- set end stop via double press of associated jog key
- do your operation away from the end stop
- single press to jog back to the set end stop

I would love to have the "smooth" movement again, but I think that having the jog segmented by the natural pitch of the current ratio makes more sense. That way we don't have to keep track of spindle sync as much.

Also apologies for not replying as much, things are hectic as normal - haha

martinlong1978, 2025-02-25:
@jh247247 

That's great. Mainly I wanted validation that I wasn't losing my mind. I'll be making efforts to get jog working in the coming days. The workflow looks fine, but the only thing that bothers me is powering up to a shoulder to set the left end stop. This is where a smooth jog might be more practical. Also, I do really want to be able to jog between the end stops without stopping or reversing the lathe. So there are a number of challenges to that. It would definitely require breaking sync. Being able to do a "fast return" would also be desirable. Say a long thread takes a minute or so to do a pass, being able to return to the start in seconds would be great. I'd be interested in your thoughts if you have any, otherwise I'll just get on with the implementation. 

I'm getting near to sending a pull request. I sort of have all the pieces out on the floor at the moment, and need to get it all tidied up. I've fixed a few glaring bugs with the calculations where the dimensional analysis just didn't make sense (what unit is that thing even in?), and it was either working because of an equal and opposite difference elsewhere. However, I really needed to get it on my lathe to validate it, and I'm waiting for some of the hardware to get that done.

Also, I do then need to revisit the Teensy implementation. I've neglected it somewhat while making stuff work, so I do just need to go back and cross patch some of that.