NoEngineerHere/TeensyELS #25: Why keep within 1 spindle revolution of the endstop?

martinlong1978, 2025-03-09:
Hi @jh247247 

I'm a bit confused about the following

[image]

Why within one revolution? For a large thread that could be like... 6mm PAST the actual endstop, and potentially cause a crash. Why not just stop AT the endstop, like this...

[image]

I'm doing quite a lot of reworking. I'll be asking a few questions, because there's some calcs I just can't see and being correct, but I might have missed something

Thanks

Martin

jh247247, 2025-03-09:
Hey! 
So this bit of code was intended to be a "fast start" for the spindle when it has hit an end stop.

Imagine that when you've hit an end stop, the spindle takes N rotations to stop after that point. When you reverse the spindle to move back you don't want to wait for those N rotations again, so instead we force the rotation to be within one rotation of the end stop position. This means that when we do move back, we restart within 1 rotation instead of the N rotations "past" the end stop.

I hope that made sense, I can graph it out if need be but I'm away from my computer right now

martinlong1978, 2025-03-09:
Thanks @jh247247 

Ah, I get it. So it's updating the expected position, but not actually making movements, so kind of keeps it in "sync", in its head at least. 

So I want to completely remove the need to reverse the spindle completely. As I see it, this firmware has 2 benefits over the clough42 verson: 

  - endstops allowing threading very close to a shoulder
  - sync, allowing disengagement of "virtual" half nut, which allows return without disengaging the spindle. 

For me, one of the big deals is the second point, and all the stop-starts. I just want to return the carriage to the start point and start cutting again. 

So, I'll leave that logic how it is anyway. It would be great if we can fulfil both goals and still use both ways of working.