NoEngineerHere/TeensyELS #27: Sending too many pulses

chbeeler, 2025-03-31:
I can't get the encoder and the motor to be in sync, even on the desk. My encoder and closed loop stepper were used with a NanoELS h2 before. I have set the steps per rotation settings and with my 1mm leadscrew and selecting 1mm thread pitch I would expect the motor to do one turn for each turn of the encoder. If I turn the encoder very slow (like 10s for one rotation) it matches: 

[image]

But when I turn it slightly faster I get to many pulses. 

[image]

I counted all encoder and motor pulses in the following picture and while I get 60 encoder pulses the motor was sent more than 70 pulses:

[image]

I tried commit 66e751d6e4f926c7c3f368b8867eb57cf34c2d4f and cfa3c95341f5cd7bc022dcd31a1832a275b93dc4 since in the latest commit on main the motor jumps back with every rotation and on boj-position-consumption-fix the motor jumps when switching to enable after the spindle has moved.

Maybe I'm using the UI wrong? My steps are:
1. Disable Lock
2. Switch from feed to thread mode
3. Switch to 1mm thread pitch
4. Press ELS_ENABLE_BUTTON to enable the motor
Now the encoder and motor should be in sync, or am I missing something?

jh247247, 2025-04-06:
Hey! Sorry to get back to you so long after you raised this issue - I think this may be related to the ratio being set incorrectly in my code, initially pointed out in #20 ... @martinlong1978 should hopefully have a PR to merge all his changes into this repo soon, but I see he's still actively working on changes here: https://github.com/NoEngineerHere/TeensyELS/compare/main...martinlong1978:TeensyELS:ESP32Support

In the meantime you can either try out his fork, or try inverting the ratio of your gearbox (it was initially a hack in my case to get things working in time for the video, so take that as you will!) 

I'm hoping to revisit this and add more features when @martinlong1978's work is merged in, so 🤞 it happens soon!

martinlong1978, 2025-04-06:
Yeah there's quite a lot of work been going on. I'm probably going to be ready to do a pull request in the next week or so. It would be good to get some real world testing on a Teensy as all of mine has been on ESP32. It should sort out all of the sync issues, and basically now will sync in the shortest space possible. (Actually the shortest space based on the current start logic, but I also plan to tune the start position to further improve that)
 Watch for the pull request, or even give it a try from my repo if you want an early look (and report any bugs on Teensy)

jh247247, 2025-05-06:
@chbeeler if you still have your system up we've just merged in the @martinlong1978 's changes, otherwise I'll get to working on the issue you've raised when I get time!